Razor-AHRS/razor-9dof-ahrs #29: Razor 9DOF SEN-10736

mhcohen56, 2015-02-09:
I've followed the direction very carefully. The Arduino sketch runs fine, but when I try to run the "Processing" sketch, I get the following messages, and nothing appears in the Razor_AHRS_test graphics window. Could you please give me some idea of what I might be doing incorrectly?

java.lang.RuntimeException: java.lang.ArrayIndexOutOfBoundsException: 22
	at com.jogamp.common.util.awt.AWTEDTExecutor.invoke(AWTEDTExecutor.java:58)
	at jogamp.opengl.awt.AWTThreadingPlugin.invokeOnOpenGLThread(AWTThreadingPlugin.java:103)
	at jogamp.opengl.ThreadingImpl.invokeOnOpenGLThread(ThreadingImpl.java:206)
	at javax.media.opengl.Threading.invokeOnOpenGLThread(Threading.java:172)
	at javax.media.opengl.Threading.invoke(Threading.java:191)
	at javax.media.opengl.awt.GLCanvas.display(GLCanvas.java:541)
	at processing.opengl.PJOGL.requestDraw(PJOGL.java:688)
	at processing.opengl.PGraphicsOpenGL.requestDraw(PGraphicsOpenGL.java:1651)
	at processing.core.PApplet.run(PApplet.java:2256)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.lang.ArrayIndexOutOfBoundsException: 22
	at Razor_AHRS_test.setup(Razor_AHRS_test.java:150)
	at processing.core.PApplet.handleDraw(PApplet.java:2361)
	at processing.opengl.PJOGL$PGLListener.display(PJOGL.java:862)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:665)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:649)
	at javax.media.opengl.awt.GLCanvas$10.run(GLCanvas.java:1289)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1119)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:994)
	at javax.media.opengl.awt.GLCanvas$11.run(GLCanvas.java:1300)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$200(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

max246, 2015-04-06:
have you changed anything on the test file? Because line 150 is a Serial.write
https://github.com/ptrbrtz/razor-9dof-ahrs/blob/master/Processing/Razor_AHRS_test/Razor_AHRS_test.pde#L150

mhcohen56, 2015-04-06:
No I don't believe that I've changed anything in the test file.  Should line 150 NOT be a Serial.write?  What should it be?
Thanks!

max246, 2015-04-07:
Ok, the problem is this one:   String portName = Serial.list()[SERIAL_PORT_NUM];
Probably your list of Serial ports is zero and it returns that error about index.

I just compiled for Linux and it works, which OS do you run it?

Jimwaga, 2015-04-07:
You change the ‎[SERIAL_PORT_NUM] to reflect your specified port number                                                                                                                                                                                                                                                                                                                                         Sent from my BlackBerry 10 smartphone.                                                                                                                                                                                                                From: max246Sent: Tuesday, 7 April 2015 15:41To: ptrbrtz/razor-9dof-ahrsReply To: ptrbrtz/razor-9dof-ahrsSubject: Re: [razor-9dof-ahrs] Razor 9DOF SEN-10736 (#29)Ok, the problem is this one:   String portName = Serial.list()[SERIAL_PORT_NUM];
Probably your list of Serial ports is zero and it returns that error about index.

I just compiled for Linux and it works, which OS do you run it?

max246, 2015-04-07:
Nope, it just worked.
Probably you got no serial port or you user dont have access to the serial port without root permissions.

Which Os are you running the script?

mhcohen56, 2015-04-07:
I am using Windows 7 - 64bit.  I've also used Linux Mint v17.  I have
selected a valid COM port.  The ascii data can be seen on the serial
monitor, but the graphics show nothing.

-- 
_________________

Marc Cohen, Ph.D.

Assistant Research Scientist

Institute for Systems Research

University of Maryland

College Park, MD 20742
o: 301.384.1115
c: 301.509.7394
e: mhcohen@umd.edu

max246, 2015-04-07:
You have to understand that you need to list your serial list ports, try to print this:
println(Serial.list());

and you will see what you get from it.

mhcohen56, 2015-04-07:
Hi max246, I'll give it a try.  Thank you.  I'm busy writing a proposal
now, so I'll try it in the next couple of days. - Marc.

-- 
_________________

Marc Cohen, Ph.D.

Assistant Research Scientist

Institute for Systems Research

University of Maryland

College Park, MD 20742
o: 301.384.1115
c: 301.509.7394
e: mhcohen@umd.edu