Razor-AHRS/razor-9dof-ahrs #33: Connecting razor IMU Directly with MEGA 2560/UNO without FTDI Basic Breakout Board

michaelcheng1991, 2015-07-04:
Hi Guys! I've connected the IMU with MEGA in the following configuration, but I kept getting 0 for all the output. Could anyone please give me some hint on how to resolve this problem?
Also, I've been trying to figure out where in the Razor_AHRS script calls the Sensors script? and is it possible for me to just use the Sensor script to get the output using MEGA without the FTDI? 

This is my first time posting question, so please let me know if I should be more specific or informative. I really appreciate if anyone could help me out! Thanks! 

IMU     |        MEGA 2560
GND  ----        GND
3.3v   -----        3.3v 
TX0    ----        RX0
RX1   ----         TX1

ThotAlion, 2015-07-13:
Hello,

I had this issue connecting RAZOR on Linux. It seems that the "zero" character is sent several times and enters the buffer only for the first line. I advise you to make a readline and remove all zero characters form the string you obtained.
I have not this issue on Windows.

michaelcheng1991, 2015-07-15:
Thank you very much for your reply ThotAlion!

mongy910, 2016-07-07:
Did that fix your issue?