Razor-AHRS/razor-9dof-ahrs #38: only getting RPY values

w4zir, 2015-10-28:
I am running Razor-9dof-ahrs with ros but I am only getting RPY values and no acceleration or gyro  values. Are there some parameters that I need to change so I will start getting values. The program ends with the following error which is because the word only has 3 elements i.e. RPY.

imuMsg.linear_acceleration.x = -float(words[3]) * accel_factor
IndexError: list index out of range

olmerg, 2015-10-28:
you should have to change the configuration in the file of the arduino:

https://github.com/ptrbrtz/razor-9dof-ahrs/blob/master/Arduino/Razor_AHRS/Razor_AHRS.ino#L177-L189

w4zir, 2015-10-29:
oh OK. I was expecting it to print all the 9 sensor values in a single line as the read in the code for ROS node, but AHRS is printing it in 3 separate lines so I had to modify code accordingly.  Thanks for help.

closed 2016-11-22 by dennisguse