Razor-AHRS/razor-9dof-ahrs #42: Building an AHRS 9DOF RAZOR tuto error #YPR=0.00,0.00,0.00 NEED HELP!!!

zelda23, 2015-12-10:
Hello,
I have problem with my RAZO 9DOF.
I try to follow this tuto:

https://github.com/ptrbrtz/razor-9dof-ahrs/wiki/Tutorial

But there is a technical problem even with the arduino UNO ( i had problem too with FTDI 3.3 v)

STEP A: "" Go to "File" and hit "Upload to I/O Board". After a short while at the bottom of the Arduino code window it should say "Done uploading".""
[image]
[image]
[image]

Until here (STEP A) is ok with arduino UNO, it is written : "Done Uploading". But after

( STEP B) when I want to test by staring at numbers, bringing up the Serial Monitor of Arduino under "Tools" → "Serial Monitor". Set it to 57600 baud, then I get only: #YPR=0.00,0.00,0.00

What is the problem? You can see files attached with pictures.

Thanks a lot

Best regards

mlvo83, 2016-11-26:
having the same issue. Did you ever get this working?

zelda23, 2016-11-26:
hello no it didn't work. I used an another AHRS of adafruit with arduino UNO.
it worked good and it was chaep.
----Message d'origine----

mlvo83, 2016-11-27:
Thanks for replying. Would you be willing to share the name of the one you used.  I have been looking at some of the adafruit sensors but I'm now getting gun shy

zelda23, 2016-11-27:
hi: "
Adafruit 10-DOF IMU Breakout - L3GD20H + LSM303 + BMP180"
https://www.adafruit.com/products/1604
----Message d'origine----

zelda23, 2016-12-19:
Thank you github

closed 2016-12-19 by zelda23