Razor-AHRS/razor-9dof-ahrs #44: Getting no GYRO values

LadAussie, 2016-02-26:
Hi all, whilst trying to calibrate the SparkFun "9DOF Sensor Stick" version "SEN-10724" I get only zeros for the gyro x,y, z values. Can anyone suggest why this may be and or a possible solution?

pdeschulthess, 2016-03-05:
I've managed to get this library (mostly) working with that board.  Tell me a little more and maybe I can help.  

1. Have you configured the firmware to use the SEN-10724?
2. Can you get text and binary readings on the serial output?
3. Do you get the #YPR lines in the serial monitor?

LadAussie, 2016-03-06:
1. Yes
2.Yes
3.Yes

everything seems to be in working order, just get only 0 for the output of the gyro. I have tried another stick also with the same issue. Its not hugely important as i am just using it to measure accelerations. Just thought it was a bit odd.

LadAussie, 2016-03-22:
Got a Razor so now I have no issue with the gryo values.