Razor-AHRS/razor-9dof-ahrs #48: Getting avrdude: stk500_getsync(): not in sync: resp=0x00 while using FTDI on a razor board without firmware

DigitalGabriele, 2016-04-21:
Hello, 

Just as the title claims, I have followed all the instructions and when I tried to upload the verified code to Arduino, it would keep on complaining about "avrdude: stk500_getsync(): not in sync: resp=0x00". Furthermore, it does not have any firmware on it. There is a chance that it may have been burnt but Id like to try rule out other issues too. 

Has anyone ever faced this issue?

maharshi114, 2019-01-01:
Yes! After using approximately four-month, suddenly the serial data not coming on Serial port. I upload the default code again and showing me the same problem.  

Also, I upload Arduino Blink code but showing me the same problem. 

@DigitalGabriele your problem is solved? 

Thank you.

SoorajR-ai, 2024-02-21:
Does this problem solved?

DigitalGabriele, 2024-03-01:
No, I haven't found a solution. I still keep getting the same problem.