Razor-AHRS/razor-9dof-ahrs #57: SEN 14001 compatibility

marscar, 2017-02-16:
Is this package compatible with the latest version (SEN 14001)?

tronicgr, 2017-03-01:
I double that request. Just got the M0 version (SEN 14001) and was looking for an easy way to adapt it for AHRS. I like that it also has SD datalogging to examine later flight issues.

mathdexter, 2017-03-24:
I triple that request. :)

bolduc-drew, 2017-04-02:
Make that a quadruple request!

ZUOHJ016, 2017-05-15:
I have the SEN-14001,It's not compatible to the Razor_AHRS (version 1.4.2)
Waiting for the newest Razor_AHRS ..........

bpinaya, 2017-06-04:
I quintuple that request! :+1:

tanyax, 2017-06-09:
So do I!!

Murphy3220, 2017-09-21:
I have updated the Razor AHRS to work with SEN-14001.  All the credit goes to the original coder.  You can find the updated code at https://github.com/Murphy3220/Razor-AHRS-SEN-14001 .  It only works for the SEN-14001.

RijadAlisic, 2017-09-30:
I have a pull request fixing this issue!

lebarsfa, 2017-11-24:
An alternate version that works with the SEN-14001 as well as previous versions, with optional ROS-compatible output mode : https://github.com/lebarsfa/razor-9dof-ahrs .

PratibhaT, 2018-04-04:
How Do i mount this sensor: SEN-14001 on my robot?
The gyro and accelerometer axis is different from that of magnetometer axis?

pulkitgoyal1506, 2018-04-04:
I am facing the problem, it's little confusing!!
https://github.com/Murphy3220/Razor-AHRS-SEN-14001
I am not sure, How Do I mount it on my robot as the axis for Magnetometer is different from that of acc and gyro?

lebarsfa, 2018-04-04:
You can consider that the top is where "sparkfun" is written (the bottom is where there are the LEDs) and the nose is where "9DOF Razor IMU" is written (the rear is where there is the micro USB port). Note however that the angle conventions might be different depending on the firmware you use on it, e.g. for firmware on https://github.com/lebarsfa/razor-9dof-ahrs (North-East-Down convention) :

- X axis pointing forward
- Y axis pointing to the right
- Z axis pointing down

While for the ROS node https://github.com/KristofRobot/razor_imu_9dof (North-West-Up convention) :

- X axis pointing forward
- Y axis pointing to the left 
- Z axis pointing up 

Another thing to check when you mount it is to put it away from the motors or other elements that might cause magnetic disturbances, sometimes it can be a good idea to mount it on top of a dedicated mast of around 20 cm if you need to rely a lot on magnetometers.
So if you are unsure, the best is to test if you get the expected values on your robot and make a coordinate system conversion if you are not happy with the conventions of the firmware you have chosen.

lebarsfa, 2018-04-04:
Normally the different orientations of the magnetos vs accel/gyros is supposed to be correctly taken into account in that firmware. Ensure also that you used the recommended version https://github.com/lebarsfa/SparkFun_MPU-9250-DMP_Arduino_Library when building the firmware. Most of the time, bad yaw values are due to a bad magnetometer calibration, ensure that you did it in a place where there are no dynamic magnetic disturbances (e.g. outside in the middle of a field)...