Razor-AHRS/razor-9dof-ahrs #59: Nano+GY 85   via usb

lukaszgrochola, 2017-06-01:
Hello!  

 I made setup for head tracking. Nano with usb interface and gy-85. So i've upload sketch, tested in processing and movement is rly good but...  When i try use it in opentrack/facetrack no ir, selecting all of the options etc. I have no movement in this programms.(in section about i see numbers running in, so the programms are getting data).  

Any ideas whats wrong?

GFA-SkyRanger, 2017-08-22:
I have exact the same problem.
On testings all works like a charm. but opentracker and facetracknoir wont get any datas with "Hatire Arduino".

GFA-SkyRanger, 2017-08-22:
ok got it to work.
#define OUTPUT__STARTUP_STREAM_ON false
has set to true.
so just calibrate axes and configure it.

SebastianBonn, 2017-09-20:
If you want to use GY-85, uncomment the following line in Razor_AHRS.ino:

#define HW__VERSION_CODE 10724 // SparkFun "9DOF Sensor Stick" version "SEN-10724" (HMC5883L magnetometer)

The only difference between SEN-10724 and GY-85 is the orientation of the HMC5883L (magnetometer) in relation to the orientation of accelerometer and gyro. This breaks the tilt compensation in Compass.ino, as the Magnetometer axes are differently orientated. A quick and dirty fix would be rotating the coordinate system in Compass.ino, for example as follows:

   float x_new = magnetom[1];
   float y_new = -magnetom[0];
   float z_new = magnetom[2];
  
  // Tilt compensated magnetic field X
  mag_x = x_new * cos_pitch + y_new * sin_roll * sin_pitch + z_new * cos_roll * sin_pitch;
  // Tilt compensated magnetic field Y
  mag_y = y_new * cos_roll - z_new * sin_roll;