Razor-AHRS/razor-9dof-ahrs #60: Obsolete sensors only supported

seamaster101, 2017-07-12:
Hi I just went to the Sparkfun site and checked all of the sensors listed:
* Razor AHRS Firmware v1.4.2
* 9 Degree of Measurement Attitude and Heading Reference System
* for Sparkfun "9DOF Razor IMU" (SEN-10125 and SEN-10736)
* and "9DOF Sensor Stick" (SEN-10183, 10321 and SEN-10724)

SEN-10125; SEN-10736; SEN-10183, SEN-10321 and SEN-10724 are all retired. No stock, no price.
there are newer breakout boards with single chip like MPU9250. For example SEN-13762.
unfortunately it contains different sensors: the MPU-6500, which contains a 3-axis gyroscope as well as a 3-axis accelerometer, and the AK8963, which features a 3-axis magnetometer.
is there a plan to update the code to support MPU9250?

a

GFA-SkyRanger, 2017-08-22:
I just got a GY-85 on ebay, its still compatible with SEN-10736.
so what?

seamaster101, 2017-08-23:
I don't quite get the "so what" part...
My question was about support for newer chips such as MPU9250.  Are there plans or not? I'm not interested in the old chips such as SEN-10736.