Razor-AHRS/razor-9dof-ahrs #73: Processing Code Not Compatible with Latest EJML

cemersoz, 2019-04-27:
Took me a while to figure this out, putting this note here in case it helps anyone. The Extended Magnetometer Calibration code for processing depends on [EJML](http://ejml.org/wiki/index.php?title=Download). As of April 2019 They are on version 0.38 which is not compatible with the processing code provided. I had to go back and use version 0.30 that you can download [here](https://sourceforge.net/projects/ejml/files/v0.30/ejml-v0.30-libs.zip/download).

Hope this helps someone!

boheyse, 2019-05-07:
Which .jar file did you use in the zip download?

cemersoz, 2019-05-14:
I just dragged and dropped all of them, that worked. Idk what specifically is needed.

JeffSpaniol, 2019-07-06:
That didn't work for me... Could you please, specify the folder yout droped it? It says that the package “org.ejml” does not exist.